moveit/warehouse_ros_mongo #5: Missing dependency mongodb-dev

davetcoleman, 2016-10-06:
Travis is failing for this repo because of a missing rosdep rule for "mongodb-dev" in xenial. Does someone have any insights? @ugocupcic?

ugocupcic, 2016-10-08:
I must admit I didn't check the rosdeps. But I think that mongodb-dev shouldn't be installed on Xenial (if it exists) as it's replaced by the mongocxx manual install step. The rosdep can probably be just dropped?

> 
> Travis is failing for this repo because of a missing rosdep rule for "mongodb-dev" in xenial. Does someone have any insights? @ugocupcic?

davetcoleman, 2016-10-08:
i can quickly merge in a PR for this :)

ethanfowler, 2016-12-06:
I've just run into the same issue on kinetic - warehouse_ros_mongo's package.xml specifies a dependency on mongodb-dev, which doesn't exist for jade or kinetic (see https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L3137). Shall I create a PR @davetcoleman?

davetcoleman, 2016-12-06:
@ethanfowler yes please!

ethanfowler, 2016-12-07:
Created PR #8.

PR #8 (2016-12-12): Fixes for xenial jade/kinetic installations
Should fix #6, failing make on kinetic docker image.

Closes #5 .
closed 2016-12-12 by davetcoleman