moveit/warehouse_ros_mongo #6: Experiemental Docker with mongo support broken

davetcoleman, 2016-10-18:
From https://hub.docker.com/r/moveit/moveit/builds/bybu65qy9524f3cazmfftxm/

Something is wrong with the Docker build setup by @ugocupcic

```
Errors << warehouse_ros_mongo:cmake /root/ws_moveit/logs/warehouse_ros_mongo/build.cmake.000.log
CMake Error at /opt/ros/jade/share/catkin/cmake/catkin_package.cmake:291 (message):
  catkin_package() absolute include dir
  '/root/ws_moveit/devel/.private/warehouse_ros_mongo/include' does not exist
Call Stack (most recent call first):
  /opt/ros/jade/share/catkin/cmake/catkin_package.cmake:98 (_catkin_package)
  CMakeLists.txt:24 (catkin_package)

cd /root/ws_moveit/build/warehouse_ros_mongo; catkin build --get-env warehouse_ros_mongo | catkin env -si  /usr/bin/cmake /root/ws_moveit/src/warehouse_ros_mongo --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/root/ws_moveit/devel/.private/warehouse_ros_mongo -DCMAKE_INSTALL_PREFIX=/root/ws_moveit/install; cd -
```

I will likely disable this build if no one is interested in fixing it, to stop the emails I'm getting.

ugocupcic, 2016-10-18:
I'll try to take a look in the next few days 

Sent from a tiny keyboard, excuse the brevity.

> From https://hub.docker.com/r/moveit/moveit/builds/bybu65qy9524f3cazmfftxm/
>
> Something is wrong with the Docker build setup by @ugocupcic
>
> Errors << warehouse_ros_mongo:cmake [...]

ethanfowler, 2016-12-07:
This was an issue in the cmake, including a folder before creating it. Should be fixed in #8.

PR #8 (2016-12-12): Fixes for xenial jade/kinetic installations
Should fix #6, failing make on kinetic docker image.

Closes #5 .
closed 2016-12-12 by davetcoleman