moveit/warehouse_ros_mongo #7: Tests broken

davetcoleman, 2016-10-22:
When building this package's tests this error occurs: https://gist.github.com/davetcoleman/f5dddabaa85c373f61cd2a906170a956

christian-rauch, 2018-10-27:
This issue should be resolved. Since commit 68fc7fc33a1ce6662a4d0607ccbd0d04ed57b18e the tests build successfully.
@rhaschke Can we close this?

rhaschke, 2018-10-27:
They are still not executed on Travis. Hence, they don't yet act as unittests.
I need some more hints how to setup a mongodb server via cmdline.

rhaschke, 2019-08-14:
> I need some more hints how to setup a mongodb server via cmdline.

@christian-rauch, could you help on this?

christian-rauch, 2019-08-14:
I lost track of this issue. The tests are not executed because MongoDB ist missing in the Docker container? Is there a warning or info message about this in the CI log?

At least on melodic, I just installed packages `mongodb-server` and `mongodb-clients` via apt. The manual installation of `mongo-cxx-driver` via the `travis.before_script` script was not required.

rhaschke, 2019-08-14:
Installing packages works fine, I think. Does this automatically run the server as well?

christian-rauch, 2019-08-14:
I think Debian autostarts services after package installation. Also, I don't recall that I manually started the service.

rhaschke, 2019-08-14:
Ok. Then I need to have a look at the error once again...

rhaschke, 2019-08-14:
I think I found the culprit:
```
Setting up mongodb-server (1:3.6.3-0ubuntu1.1) ...
invoke-rc.d: could not determine current runlevel
invoke-rc.d: policy-rc.d denied execution of start.
```
Looks like `mongodb-server` is not automatically started within the docker container.
I'm working on a solution.

rhaschke, 2019-08-20:
Fixed via #27.

closed 2019-08-20 by rhaschke

christian-rauch, 2019-08-20:
How was this fixed in the end if #27 is closed without merging?

rhaschke, 2019-08-20:
I manually merged #27 and github didn't noticed.