moveit/warehouse_ros_mongo #14: Use released libmongocxx_ros

christian-rauch, 2018-01-23:
The MongoDB C++ driver has been released as part of kinetic (package `ros-kinetic-libmongocxx-ros`). I can successfully build `warehouse_ros_mongo` using this binary package instead of using the `26compat` version from source. I therefore assume that the APIs of both versions are compatible.

However, the plugin crashes once I want to connect to the database via the MoveIt! RViz Plugin:

```
rviz: /usr/include/boost/smart_ptr/scoped_ptr.hpp:99: T* boost::scoped_ptr<T>::operator->() const [with T = mongo::AtomicWord<unsigned int>]: Assertion `px != 0' failed.

Thread 17 "rviz" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fff89dc5700 (LWP 2888)]
0x00007ffff69dc428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff69dc428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff69de02a in __GI_abort () at abort.c:89
#2  0x00007ffff69d4bd7 in __assert_fail_base (fmt=<optimised out>, assertion=assertion@entry=0x7fff7b67aec4 "px != 0", file=file@entry=0x7fff7b67ae98 "/usr/include/boost/smart_ptr/scoped_ptr.hpp", 
    line=line@entry=99, function=function@entry=0x7fff7b67af40 "T* boost::scoped_ptr<T>::operator->() const [with T = mongo::AtomicWord<unsigned int>]") at assert.c:92
#3  0x00007ffff69d4c82 in __GI___assert_fail (assertion=0x7fff7b67aec4 "px != 0", file=0x7fff7b67ae98 "/usr/include/boost/smart_ptr/scoped_ptr.hpp", line=99, 
    function=0x7fff7b67af40 "T* boost::scoped_ptr<T>::operator->() const [with T = mongo::AtomicWord<unsigned int>]") at assert.c:101
#4  0x00007fff7b5b50b7 in mongo::OID::Increment::next() () from /opt/ros/kinetic/lib/libmongoclient.so
#5  0x00007fff7b5b555e in mongo::OID::init() () from /opt/ros/kinetic/lib/libmongoclient.so
#6  0x00007fff7b60a168 in ?? () from /opt/ros/kinetic/lib/libmongoclient.so
#7  0x00007fff7b60aa97 in ?? () from /opt/ros/kinetic/lib/libmongoclient.so
#8  0x00007fff7b5cc59c in mongo::DBClientBase::insert(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<mongo::BSONObj, std::allocator<mongo::BSONObj> > const&, int, mongo::WriteConcern const*) () from /opt/ros/kinetic/lib/libmongoclient.so
#9  0x00007fff7b5ca5f5 in mongo::DBClientBase::insert(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, mongo::BSONObj, int, mongo::WriteConcern const*) ()
   from /opt/ros/kinetic/lib/libmongoclient.so
#10 0x00007fff882179cb in warehouse_ros_mongo::MongoMessageCollection::initialize (this=0x7fff8000aa40, datatype="moveit_msgs/PlanningScene", md5="89aac6d20db967ba716cba5a86b1b9d5")
    at /home/christian/Development/kuka_ws/src/warehouse_ros_mongo/src/message_collection.cpp:72
#11 0x00007fff97b4aab9 in warehouse_ros::MessageCollection<moveit_msgs::PlanningScene_<std::allocator<void> > >::MessageCollection(boost::shared_ptr<warehouse_ros::MessageCollectionHelper>) ()
   from /opt/ros/kinetic/lib/libmoveit_warehouse.so.0.9.11
#12 0x00007fff97b51f94 in warehouse_ros::MessageCollection<moveit_msgs::PlanningScene_<std::allocator<void> > >::Ptr warehouse_ros::DatabaseConnection::openCollectionPtr<moveit_msgs::PlanningScene_<std::allocator<void> > >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /opt/ros/kinetic/lib/libmoveit_warehouse.so.0.9.11
#13 0x00007fff97b44aad in moveit_warehouse::PlanningSceneStorage::createCollections() () from /opt/ros/kinetic/lib/libmoveit_warehouse.so.0.9.11
#14 0x00007fff97b44d7c in moveit_warehouse::PlanningSceneStorage::PlanningSceneStorage(boost::shared_ptr<warehouse_ros::DatabaseConnection>) () from /opt/ros/kinetic/lib/libmoveit_warehouse.so.0.9.11
#15 0x00007fffac905bd9 in moveit_rviz_plugin::MotionPlanningFrame::computeDatabaseConnectButtonClicked() () from /opt/ros/kinetic/lib/libmoveit_motion_planning_rviz_plugin_core.so.0.9.11
#16 0x00007fff976d183b in moveit::tools::BackgroundProcessing::processingThread() () from /opt/ros/kinetic/lib/libmoveit_background_processing.so.0.9.11
#17 0x00007ffff60f85d5 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
#18 0x00007ffff5ed16ba in start_thread (arg=0x7fff89dc5700) at pthread_create.c:333
#19 0x00007ffff6aae41d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

Is there any chance that `warehouse_ros_mongo` will work with `libmongocxx_ros`? This would enable `warehouse_ros_mongo` to be binary released into the kinect repo.

rhaschke, 2018-10-26:
Enabling a binary release of this package would be great! Please provide a PR.

christian-rauch, 2018-10-27:
Since Ubuntu 18.04 you only need to install the package `libmongoclient-dev` to build. After this, `warehouse_ros_mongo` is building out of the box.

Since this is already in the osdeps as [`libmongoclient-dev`](https://github.com/ros/rosdistro/blob/2b1425658a60cba45b28f71d8e658c79fb787dc7/rosdep/base.yaml#L2279), the original release package jsut needs to be re-released for meldic.

rhaschke, 2018-10-27:
Addressed in #19. Thanks for pointing this out!

closed 2018-10-27 by rhaschke