moveit/warehouse_ros_mongo #22: cannot connect to database through moveit

azompas, 2019-01-22:
Hello,

I am using warehouse_ros_mongo on jade-devel branch and ROS kinetic. When I am trying to connect to the database through moveit I cannot and I get the following error

[ERROR] [1547208584.801465835, 54.896000000]: backgroundJob already running: PeriodicTaskRunner

I figured out that the following commit causes the error above

https://github.com/ros-planning/warehouse_ros_mongo/commit/cd9a32e2d6a0b43951c9173933d6f81ad6ca9dca

Would it be possible to revert this commit?

VitaminLL, 2020-07-15:
Same question on Franka panda arm.

rhaschke, 2020-07-20:
Try building the `melodic-devel` branch. I added a corresponding Travis job for Kinetic and it works w/o problems:
https://travis-ci.com/github/ros-planning/warehouse_ros_mongo/builds/176290761

Follow the instructions [here](https://github.com/ros-planning/warehouse_ros_mongo#building-from-source) to build the mongo C++ client lib on Xenial.

closed 2020-07-20 by rhaschke