moveit/warehouse_ros_mongo #31: Dependency on mongodb-server-core

simonschmeisser, 2020-04-07:
I tried following the benchmark tutorial but connecting to the database failed as it was not installed. Is there a reason why there is no dependency on `mongodb-server-core`?

v4hn, 2020-04-07:
> Is there a reason why there is no dependency on `mongodb-server-core`?

Technically you could have the server running on a different machine, couldn't you?

If you are looking into the warehouse infrastructure, you might want to try
https://github.com/k-okada/warehouse_ros_sqlite unless you are bound to use mongodb.

simonschmeisser, 2020-04-07:
I'm just trying to follow the tutorial for the benchmark to get some progress in my assistant PR ...

but if we have a sqlite backend we should promote that in the tutorials!

v4hn, 2020-04-07:
> but if we have a sqlite backend we should promote that in the tutorials!

Well, nobody tested it really. :) Thus also nobody wrote tutorials.
@k-okada implemented the plugin a while ago.

rhaschke, 2020-05-20:
I agree with @v4hn, that the database should be run separately. I think the documentation should be improved in the tutorials. Closing here.

closed 2020-05-20 by rhaschke