moveit/warehouse_ros_mongo #33: Port to windows

kokkalisko, 2020-09-29:
I want to use warehouse database to build and store a planning scene with MoveIt Motion Planning in RViz for Windows. For this purpose I want to use the plugin/implementation created by this package i.e. `warehouse_ros_mongo::MongoDatabaseConnection` and in order to do so I need to build the package. Of course this is not possible in Windows, because one of its dependencies is MongoDB, whose source code is only available (as far as I know) for Linux.

In short, my questions are: 

1. has `warehouse_ros_mongo` been ported to Windows? 

2. is there an alternative implementation of warehouse_ros for mongo on Windows that I do not know of?
 
3. if none of the above exists is there any other alternatives on how to store my planning scene?

felixvd, 2020-09-29:
1., 2. Not that I know of, but I do not keep track

3. You can call the `getPlanningScene` capability of your move_group and store the `moveit_msgs::PlanningScene` object it returns, for example as a YAML file.

kokkalisko, 2020-10-01:
> 1., 2. Not that I know of, but I do not keep track
> 
> 1. You can call the `getPlanningScene` capability of your move_group and store the `moveit_msgs::PlanningScene` object it returns, for example as a YAML file.

That's what I thought. I was hoping for something that enables to use fully the Motion Planning plugin functionality for RViz and not make it from scratch. 

This plugin enables loading, storing and deleting of single objects and scenes in total. In Ubuntu it makes very easy the creation of several scenes.

felixvd, 2020-10-01:
It should not be complicated to change the Motion Planning plugin to export to/import from a YAML file. I would welcome a PR, because it would make that functionality accessible without having to set up a DB server. The current functionality could become optional.

ooeygui, 2021-06-21:
Working on MoveIt2 for Windows.
https://github.com/ros-planning/moveit2/issues/504