moveit/warehouse_ros_mongo #37: Port to ROS2

RoboticsYY, 2020-10-10:
I'm porting this repository to ROS2. I already has a [branch](https://github.com/RoboticsYY/warehouse_ros_mongo/tree/pr-porting_to_ros2) under my fork working against ROS2 Foxy. If the owner could create a `ros2` branch, I can submit a PR against it?

felixvd, 2020-10-10:
I'm happy to, but I don't know the policy. Since moveit2 and navigation2 have changed the repo's name, is anything decided for packages like this?

rhaschke, 2020-10-13:
@RoboticsYY, I have pushed a `ros2` branch.

closed 2020-10-13 by rhaschke

RoboticsYY, 2020-10-13:
@rhaschke Thanks!

@felixvd To my observation, I think only moveit2 and navigation2 changed the repo's name. I guess this is to highlight that the whole code base of the big projects are based on a new ROS. I think other packages still follow the conversion that use a ROS2 branch.