moveit/warehouse_ros_mongo #55: ROS2 - error when building warehouse_ros_mongo

nkalupahana, 2021-05-03:
When I try to build `warehouse_ros_mongo` (ROS2), I get the following error:
```
--- stderr: warehouse_ros_mongo
/Users/nisala/ws_moveit2/src/warehouse_ros_mongo/src/message_collection.cpp:83:10: error: no member named 'ensureIndex' in 'mongo::DBClientConnection'
  conn_->ensureIndex(ns_, BSON(field << 1));
  ~~~~~~~^
```

I'm using the `mongo-cxx-driver` from the `releases/legacy` branch (https://github.com/mongodb/mongo-cxx-driver/tree/releases/legacy), which I also built and installed from source. Any ideas as to what's going on? Thanks!

- ROS Distro: Foxy
- OS Version: macOS 11

tylerjw, 2021-05-03:
I believe that is defined in the libmongoclient-dev package in ubuntu focal.  Here is the link to the package from ubuntu.  Maybe there is a specific tag you need to use:

https://packages.ubuntu.com/focal/libmongoclient-dev

nkalupahana, 2021-05-03:
Ah, okay. Let me try building from the source files they have for the package, and I'll get back to you. Thanks!

tylerjw, 2021-05-03:
There is also a chance it comes from this package (but I believe that is less likely): https://packages.ubuntu.com/focal/mongodb

One thing to note is that we have a custom cmake macro for finding the mongodb package here: https://github.com/ros-planning/warehouse_ros_mongo/blob/ros2/cmake/FindMONGODB.cmake

There is a chance some of those paths are ubuntu specific and you'll need to adjust them for your macOS source build.

nkalupahana, 2021-05-03:
Found the error:
http://mongocxx.org/legacy-v1/breaking-changes/
> The ensureIndex and related methods have been removed. The replacement is the new createIndex method.

Looks like you have to use `26compat`, the legacy version of the v1 legacy driver.

Are there any plans to move to the latest 3.x driver? This driver is super old, and just to build and install it I have to relink my system with Python 2, which has reached EOL.

tylerjw, 2021-05-03:
> The 3.x driver, on the other hand, is much more modern and is already on package managers like `brew`. I also had to go in and edit the driver code because the version of boost it uses is so old, which is just not ideal.

We would happily accept pull requests for this.  I don't know of any one else working on this.  One thing to consider though is that for now we are limited to what is packaged for ubuntu 20.04 (on foxy, and currently on rolling).  There are probably newer interfaces packaged for ubuntu, we would just need to make sure the keys are in rosdep so they can be installed that way so the buildfarm and ros release for this works.

nkalupahana, 2021-05-07:
Hm, alright. I'm unfortunately deep in finals season at university, but maybe I'll look into this over the summer.

closed 2021-05-07 by nkalupahana