moveit/warehouse_ros_mongo #66: Migrate to 3.x API

ooeygui, 2021-06-21:
The 2.6 compatibility layer for Mongo is EOL.

To MoveIt1 and MoveIt2 on Windows, the 3.x version if the MongoDB would be required.

Tobias-Fischer, 2021-08-09:
+1 for this - we will soon have libmongocxx in conda-forge (https://github.com/conda-forge/staged-recipes/pull/15867) and could package warehouse-ros-mongo in RoboStack (https://github.com/RoboStack/ros-galactic/) if the API is migrated to 3.x. Putting the legacy driver in conda-forge seems tricky - it's hugely outdated ..

@ooeygui - do you have any plans for the migration?

/cc @wolfv @traversaro

v4hn, 2021-08-09:
Personally, I would prefer to deprecate warehouse-ros-mongo altogether in favor of https://github.com/gleichdick/warehouse_ros_sqlite .

However, if anyone provides a patch for the Mongo API migration, we will be happy to accept it.

ooeygui, 2021-08-09:
@v4hn I like this idea as well; What would the process be to "adopt" the warehouse_ros_sqlite into ros-planning?

@Tobias-Fischer We have had commercial requests to support this **kind** of functionality, but not necessarily for Mongo. SQL-lite might be the best solution. I haven't specifically investigated sqlite support though - I'll put it on the backlog.

Tobias-Fischer, 2021-08-09:
The SQLite solution also sounds good to me!

v4hn, 2021-08-10:
> @v4hn I like this idea as well; What would the process be to "adopt" the warehouse_ros_sqlite into ros-planning?

It's basically on the way and was discussed in the maintainer meeting recently. https://github.com/gleichdick/warehouse_ros_sqlite/issues/3