moveit/warehouse_ros_mongo #69: Problems with building on Nvidia Jetson

Serafadam, 2021-10-06:
Hi, I tried to build this package on Nvidia Xavier Jetson (in my Galactic Docker container), and there were some errors:

```
--- stderr: warehouse_ros_mongo                                                                                                                                                                                 
/usr/bin/ld: CMakeFiles/test_warehouse_ros_mongo_cpp.dir/test/test_warehouse_ros_mongo.cpp.o: undefined reference to symbol '_ZN5boost6system15system_categoryEv'
//usr/lib/aarch64-linux-gnu/libboost_system.so.1.65.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [test_warehouse_ros_mongo_cpp] Error 1
make[1]: *** [CMakeFiles/test_warehouse_ros_mongo_cpp.dir/all] Error 2
make: *** [all] Error 2

```
I managed to fix it by adding some linking in CMakeLists.txt:
```
target_link_libraries(${PROJECT_NAME} ${Boost_FILESYSTEM_LIBRARY})
target_link_libraries(${PROJECT_NAME} ${Boost_SYSTEM_LIBRARY})
target_link_libraries(${PROJECT_NAME} ${Boost_THREAD_LIBRARY})
```
I can make a PR with those changes, I guess this doesn't really affect other builds.

henningkayser, 2021-10-07:
Great! Would you mind preparing a PR for this?

Serafadam, 2021-10-07:
Sure, here it is https://github.com/ros-planning/warehouse_ros_mongo/pull/70

henningkayser, 2021-10-07:
Thanks, @Serafadam!

closed 2021-10-07 by henningkayser