moveit/warehouse_ros_mongo #71: warehouse_ros_mongo in ROS 2 Rolling has not been released on Ubuntu Jammy

nuclearsandwich, 2022-02-24:
The ROS 2 Rolling Ridley distribution recently [migrated to Ubuntu 22.04 Jammy](https://github.com/ros/rosdistro/pull/32036) using the rosdistro migration tool.

This repository is one which failed to bloom during the migration because the `mongodb` dependency is not available in Ubuntu Jammy. Since this dependency is not present this package may need to be re-worked to remove the dependency or removed from Rolling.

Packages which are not released in Rolling may not be automatically added to future stable ROS 2 distributions.

kryptoniancode, 2023-04-11:
What is possible soultion for it?

```
executing command [apt-get install -y ros-humble-warehouse-ros-mongo]
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package ros-humble-warehouse-ros-mongo
ERROR: the following rosdeps failed to install
  apt: command [apt-get install -y ros-humble-warehouse-ros-mongo] failed
  apt: Failed to detect successful installation of [ros-humble-warehouse-ros-mongo]
```

130s, 2023-06-08:
As mentioned in https://github.com/ros-planning/warehouse_ros_mongo/issues/75, the pkg is now at least listed in rosdistro for `rolling` (debians are not yet available as discussed in the link).

130s, 2023-06-16:
> This repository is one which failed to bloom during the migration because the mongodb dependency is not available in Ubuntu Jammy. Since this dependency is not present this package may need to be re-worked to remove the dependency or removed from Rolling.

Ah. Steven already explained that an year ago :/
Anyways, conversation is continued in #75.