moveit/warehouse_ros_mongo #75: ROS2 Humble version missing

prasuchit, 2023-05-03:
Hi,

I'm unable to install the warehouse-ros-mongo package for humble using the command:

     sudo apt-get install -y ros-humble-warehouse-ros-mongo

I get this error:

     E: Unable to locate package ros-humble-warehouse-ros-mongo

Kindly help! :/

130s, 2023-06-07:
- [`warehouse_ros_mongo` is in humble's rosdstro](https://github.com/ros/rosdistro/blob/1baf91a8d769f23836e4e9f21e6faa5c5a4cdad3/humble/distribution.yaml#LL8032C7-L8032C7).
- In [packages.ros.org](http://packages.ros.org/ros2/ubuntu/dists/jammy/main/binary-amd64/Packages) I do see some other `warehouse-ros*` but not `warehouse-ros-mongo` listed for humble (nor for rolling).

130s, 2023-06-07:
@nuclearsandwich You may be able to help this?

- 20230320 I saw https://github.com/ros/rosdistro/pull/36536 was merged in.
- 20230426 AFAICT latest of some `ros-humble-warehouse-*` on http://packages.ros.org/ros2/ubuntu/dists/jammy/main/binary-amd64/Packages are built.

Would there be any build triggered after GBP repo URL change? If not, that might be the culprit? Just a guess though.

130s, 2023-06-08:
Missing in `rolling` too. Coincidence or not, both humble and rolling rosdistro were updated in https://github.com/ros/rosdistro/pull/36536.
@prasuchit would you mind updating the ticket's title to indicate both humble and rolling?

nuclearsandwich, 2023-06-16:
> Would there be any build triggered after GBP repo URL change? If not, that might be the culprit? Just a guess though.

There is no releasable version of these packages on Ubuntu Jammy because the mongodb dependency is missing.
As part of the transition to Ubuntu Jammy in Rolling, the release stanza for the package had its `version` field dropped, preserving source and release repository metadata but stopping the buildfarm from attempting to create and build source and binary packages on any platform.

The relevant distribution.yaml entries for Rolling: https://github.com/ros/rosdistro/blob/9fe8cabefd231fb6cf7722cfbfa179322c805cf4/roling/distribution.yaml#L6991-L6994
and Humble: https://github.com/ros/rosdistro/blob/9fe8cabefd231fb6cf7722cfbfa179322c805cf4/humble/distribution.yaml#L8090-L8093

As far as I know MongoDB is no longer available in upstream due in part to the license changes upstream, making the dependency on mongodb impossible to satisfy using the [expected rosdep sources](https://github.com/ros/rosdistro/blob/master/CONTRIBUTING.md#expected-rosdep-sources), which are in turn the only valid sources for build farm builds.

In order to make this package releaseable you'd have to remove the dependency on mongodb or find a package in Debian and Ubuntu which provides the mongodb dependency.

130s, 2023-06-16:
Indeed I [no longer see `mongodb` pkg on packages.ubuntu.com for jammy](https://packages.ubuntu.com/search?keywords=mongodb) and it's removed from `rosdep key` for jammy https://github.com/ros/rosdistro/pull/31551.

As far as I read [installation instruction on mongodb.com](https://www.mongodb.com/docs/manual/tutorial/install-mongodb-on-ubuntu/), deb packages are available but their custom `apt` key must be added, which removes `rosdep` from an option IINM.

Not sure what to do esp. as a non-maintainer.  Pinging @ros-planning/moveit-maintainers

nuclearsandwich, 2023-06-16:
> deb packages are available but their custom `apt` key must be added, which removes `rosdep` from an option IINM.

Indeed you're not mistaken. Officially released packages do not allow custom repositories to be used in dependencies which is enforced by the build farm not allowing custom repositories.