moveit/warehouse_ros_mongo #76: [Noetic] warehouse_ros_mongo fails to build on armhf

sloretz, 2023-05-23:
Hey @tylerjw,

It looks like `warehouse_ros_mongo` isn't building on some platforms:

* [Ubuntu Focal armhf](https://build.ros.org/view/Nbin_ufhf_uFhf/job/Nbin_ufhf_uFhf__warehouse_ros_mongo__ubuntu_focal_armhf__binary/)

I'm planning to add all chronically failing packages to per-platform blacklists later this week.
That just means the buildfarm won't try to build them every day.

When all issues causing the build failures are resolved, please make a PR to the `production` branch on [ros_buildfarm_config](https://github.com/ros-infrastructure/ros_buildfarm_config/tree/production/noetic) removing it from the `package_blacklist`:

* [release-armhf-build.yaml](https://github.com/ros-infrastructure/ros_buildfarm_config/blob/production/noetic/release-armhf-build.yaml)

rhaschke, 2023-05-24:
Looks like [Focal `armhf` is missing the meta-package `mongodb`](https://packages.ubuntu.com/focal/mongodb), which is listed as a dependency:
https://github.com/ros-planning/warehouse_ros_mongo/blob/e1615a3f5eb09638c1d4853b475aa31098371950/package.xml#L25

The only dependency of `mongodb` is `mongodb-server`, [which actually exists on _all_ supported platforms](https://packages.ubuntu.com/focal/mongodb-server).

@sloretz, is it possible to point the `mongodb` dependency to `mongodb-server` on armhf in rosdistro's [`base.yaml`](https://github.com/ros/rosdistro/blob/8d667705ed7db41b1e203343f69e3d4fcbb23743/rosdep/base.yaml)?

sloretz, 2023-06-08:
> is it possible to point the mongodb dependency to mongodb-server on armhf in rosdistro's [base.yaml](https://github.com/ros/rosdistro/blob/8d667705ed7db41b1e203343f69e3d4fcbb23743/rosdep/base.yaml)?

Adding `mongodb-server` to the rosdep key seems like a reasonable workaround to me. This seems like an upstream packaging issue, so I'll open a bug upstream to see if it's possible to do an SRU to add it to armhf, and in the meantime i won't blacklist this one.