moveit/warehouse_ros_mongo #78: libmoveit_warehouse.so Not found in ROS 2 Humble in Ubuntu 22.04

kavikode, 2024-10-06:
install/moveit_ros_warehouse/lib/libmoveit_warehouse.so.2.5.5: undefined reference to `warehouse_ros::DatabaseLoader::~DatabaseLoader()'
collect2: error: ld returned 1 exit status

Any suggestions please?

rhaschke, 2024-10-06:
Do you have `/opt/ros/humble/lib/libwarehouse_ros.so` from package `ros-humble-warehouse-ros` installed?

rhaschke, 2025-07-11:
@liamcarlos: If your issue was just similar (not same), you should provide a detailed error message on your own.
@kavikode: any progress on your side?
I'm not using ROS2 actively and I don't know the status of those packages in the different versions. Better an official MoveIt2 maintainer answers on this.

liamcarlos, 2025-07-11:
@rhaschke It complains about the same reference to warehouse_ros::DatabaseLoader::~DatabaseLoader(), but not the same version of /opt/ros/humble/lib/libmoveit_warehouse.so

```
--- stderr: dgpython_ros2_actionserver                                                                                                                         
/usr/bin/ld: /opt/ros/humble/lib/libmoveit_warehouse.so.2.7.4: undefined reference to `warehouse_ros::DatabaseLoader::~DatabaseLoader()'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/dgpython_action_server.dir/build.make:963: dgpython_action_server] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/dgpython_action_server.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
```